QUB-ASL/bzzz #28: ESCs not arming/functioning properly

alphaville, 2023-05-04:
The ESCs sometimes don't arm or don't function properly and then we have to reset the ESP32.

alphaville, 2023-05-04:
[This guy here](https://youtu.be/ce81bDdOn0M) is suggesting to arm by increasing the throttle *gradually* from a low value to a higher one until the motors start to spin.

alphaville, 2023-05-04:
More interestingly, [this guy](https://youtu.be/obR5Gky3sms) says that some ESCs may refuse to arm - or will behave weird - if they are powered, but it takes a long time to arm. In our `setup` function, `setupMotors` is called last. Maybe it should be first (after we beep the buzzer a couple of times to warn the operator).

jamie-54, 2023-05-05:
Hi, we could check if the first video works.
Just checking I thought we didn't want to arm the motors until the Arm switch was used?
In which case we have to wait more than 20s for the Pi script to work.
The second video seems to imply that we shouldn't have an arm switch, it should just arm when the quadrotor is switched on. 
We could do this and just work with one on-off switch?

alphaville, 2023-05-05:
Oh, yes, you're right. Moreover, the operator may simply want to arm the ESCs after some time. Did you try the first solution?

alphaville, 2023-05-09:
@jamie-54 if the problem hasn't been resolved we should ask a question at one of the following channels:

- [Electronics SE](https://electronics.stackexchange.com/)
- [r/Multicopter/](https://www.reddit.com/r/Multicopter/) on Reddit
- [RC_ESC on GitHub](https://github.com/RB-ENantel/RC_ESC?utm_source=platformio&utm_medium=piohome)

alphaville, 2023-05-11:
That's excellent - we're narrowing it down...

gunturiCM, 2023-05-31:
This has been fixed and addressed in #52 .

alphaville, 2023-06-01:
Reopening because the PR has not been reviewed/merged

gunturiCM, 2023-06-01:
This has been fixed and addressed in #52 and the fix has been approved now.

closed 2023-06-01 by gunturiCM