QUB-ASL/bzzz #29: Fail safes

alphaville, 2023-05-04:
Implement the following safety measures (these are common safety measure I've found online):

- [ ] **Arm switch:** the motors will not arm unless the arm switch is flipped
- [ ] **Kill switch:** the kill switch kills the motors
- [ ] **Buzzer:** the buzzer makes a sound before arming (or re-arming) and after disarming
- [ ] **Lost connection:** they say that if the copter looses connection, it should shutdown immediately

jamie-54, 2023-05-05:
Hi, the arm and kill switch are working but we might want to modify them.
Currently switch B Arms the motors and switch A can stop and start the motors.
The buzzer is connected to GPIO 27 so we could probably make a function for it.

alphaville, 2023-05-05:
@jamie-54 has this been pushed? I don't see anything here.

jamie-54, 2023-05-05:
yea in the `main.cpp` of the `feature/flight-zero` branch. It looks like some parts didn't merge into the `dev` and `main` branch

alphaville, 2023-05-05:
That's not possible. Actually I had a look at `feature/flight-zero` before we merged it and the fail safes weren't there. I thought you wanted to introduce them later. If you have the code locally, can you push it to a new branch out of `dev` and open a PR?

jamie-54, 2023-05-06:
Actually if we look at the [network graph](https://github.com/QUB-ASL/bzzz/network) we can see where things went a bit funny. The correct code seems to be showing in the [`main.cpp`](https://github.com/QUB-ASL/bzzz/blob/feature/flight-zero/src/main.cpp) of  the `feature/flight-zero` branch, lines 130-135 is the arming and lines 146-154 is the kill switch. I'll create a PR now and it should sort it out, thanks

closed 2023-07-06 by alphaville