QUB-ASL/bzzz #48: Possible issues with reading data from RPi

alphaville, 2023-05-24:
This is how we read the data from the Raspberry Pi:

```c++
String allDataFromPi;
if (Serial.available() > 0) {
    allDataFromPi = Serial.readStringUntil('\n');
    for (int i = 0; i < 16; i++) {
         m_channelData[i] = allDataFromPi.substring(1, allDataFromPi.indexOf(",")).toInt();
         allDataFromPi = allDataFromPi.substring(allDataFromPi.indexOf(",") + 1);
    }
}
```

There may be some issues here:

- When we start to read data from the Pi, we may start to read a line mid way, so we will not read all 16 numbers. A possible solution can be to put a certain character at the beginning of every message (e.g., `M` or something)
- It is not impossible that the Pi sends garbage data every now and then.
- We can try to increase the baud rate. The Pi supports baud rates up to 2M. The ESP32 can allegedly go up to 5M, although some users have reported issues above 2M. I would try 500,000.
- We can decrease the size of the message. We can encode the state of the switches as 0/1.

gunturiCM, 2023-05-31:
The suggestions have been adapted and tested in #52. After testing, no issues were found. Closing this issue.

alphaville, 2023-06-01:
Reopening because the PR has not been reviewed/merged

closed 2023-06-01 by gunturiCM