QUB-ASL/bzzz #58: Some tips for using multi-threading in R-Pi radio receiver code.

gunturiCM, 2023-06-01:
**Describe the solution you'd like**

1. implement the code in R-Pi as multiple threads with appropriate frequencies. Might want to use `.join()` and manage threads using `ThreadPoolExecutor` as a context manager for multi-threading tasks such as altitude hold, and loitering algos for future usage.
2.  When a thread terminates with an exception, the `ThreadPoolExecutor` will not report it, so might want to handle exceptions properly in the threaded functions.
3. The order of execution of pooled threads cannot be predicted, this is highly dependent on what the OS does at that time. This will cause many synchronization issues and racing conditions if they use shared resources, for example, let us say that `thread_1` needs results of `thread_2`, so the required order of execution is to start `thread_2`, let it finish, start `thread_1`, let it finish. But pooling threads 1 and 2 cannot guarantee which of the two runs first. It becomes worse if they share resources, as one might not be aware that the other changed some values in the shared resource (not thread-safe). A possible solution is to use Locks (but make sure to release them).

alphaville, 2023-08-22:
@gunturiCM what is the status of this issue? Should we close it?

alphaville, 2023-09-11:
@gunturiCM @jamie-54 do you need this issue? Shall we close it? By the way, if you need to keep notes somewhere, you can use the [discussions](https://github.com/QUB-ASL/bzzz/discussions/) page.

gunturiCM, 2023-09-11:
Hello sir, these notes can be useful in the future. For now, the `Scheduler` class handles these tasks. If ROS were to be used, these notes become obsolete.

closed 2023-09-11 by gunturiCM

alphaville, 2023-09-11:
@gunturiCM OK, thank you. You can put it in the discussions section.