QUB-ASL/bzzz #65: Run Python script on startup

alphaville, 2023-06-02:
Run the Python script on startup using `crontab`. Make sure you run it from within a virtual environment. Check the old Raspberry Pi (run `crontab -e`).

gunturiCM, 2023-06-05:
I followed this [guide](https://www.tomshardware.com/how-to/run-script-at-boot-raspberry-pi#:~:text=How%20to%20Run%20a%20Script%20at%20Raspberry%20Pi,executable%20if%20it%20is%20a%20Bash%20script.%20) and ran this command `sudo crontab -e` and added this line
``
@reboot python3 /home/bzzz/Desktop/bzzz/raspberry/read_sbus/read_sbus_from_GPIO_receiver.py
``
at the end of the file.

alphaville, 2023-06-05:
@gunturiCM just be careful with `sudo`. You may or may not need to use it. Does it work now?

gunturiCM, 2023-06-06:
> @gunturiCM just be careful with `sudo`. You may or may not need to use it. Does it work now?

Yes, professor, it works now.

closed 2023-06-06 by gunturiCM