QUB-ASL/bzzz #66: Yaw control from the RC

alphaville, 2023-06-03:
Let us re-introduce the following line in [`main.src`](https://github.com/QUB-ASL/bzzz/blob/main/src/main.cpp#L94)

```c++
yawReferenceRad += radio.yawRateReferenceRadSec() * SAMPLING_TIME;
```

But,

- Make sure `radio.yawRateReferenceRadSec() ` is zero unless we move the left stick; equivalently,
- `yawReferenceRad` doesn't drift unnecessarily
- Make sure the magnitude of the values of `radio.yawRateReferenceRadSec()` are reasonable (experimentation will be needed)

gunturiCM, 2023-06-07:
I have opened another _experimental_ branch to see if we can use the `yawRateReference` to control the `Yaw` and have the `yawreference = currentYaw`.

gunturiCM, 2023-06-09:
This has been addressed and solved in #71. The code was tested and successfully merged.

closed 2023-06-09 by jamie-54