QUB-ASL/bzzz #101: Branches Tidy up

jamie-54, 2023-08-17:
Branches [`feature/altitude_hold_control`](https://github.com/QUB-ASL/bzzz/tree/feature/altitude_hold_control), [`feature/simulations`](https://github.com/QUB-ASL/bzzz/tree/feature/simulations), and [`feature/43-angular_velocities_bias`](https://github.com/QUB-ASL/bzzz/tree/feature/43-angular_velocities_bias) are all closed and branch [`feature/92-SCL_trace_fix`](https://github.com/QUB-ASL/bzzz/tree/feature/92-SCL_trace_fix) has been merged. 

Is it ok if I delete these branches @gunturiCM @alphaville ?

alphaville, 2023-08-17:
We can safely delete [feature/92-SCL_trace_fix](https://github.com/QUB-ASL/bzzz/tree/feature/92-SCL_trace_fix). 

I'm not sure abou the rest. @gunturiCM do you need any of these branches?

gunturiCM, 2023-08-17:
We need them.

jamie-54, 2023-08-28:
now that we have a new release can we delete any branches not needed? @gunturiCM @alphaville

alphaville, 2023-08-30:
@jamie-54 I think @gunturiCM said that he needs some code which is in some of the branches. I suppose he is mainly referring to [`feature/Complete_altitude_hold_clean_code_version`](https://github.com/QUB-ASL/bzzz/tree/feature/Complete_altitude_hold_clean_code_version). @gunturiCM do you still need this? Should we delete it?

The following branches have been merged into [`feature/Complete_altitude_hold_clean_code_version`](https://github.com/QUB-ASL/bzzz/tree/feature/Complete_altitude_hold_clean_code_version):

- `feature/LQR_KF_Scheduler_altitude_control`
- `feature/altitude_hold_control`
- `feature/LQR_altitude_control`
- `feature/kalman_filter_altitude`

The branch [`feature/simulations`](https://github.com/QUB-ASL/bzzz/compare/feature/simulations) is very old and I wouldn't approve the changes it contains, so I'd vote to delete it.

The branch `feature/43-angular_velocities_bias` is associated with #43, which has been closed, so I'd vote to delete the branch as well.

@gunturiCM any objections?

gunturiCM, 2023-08-30:
- I agree merging `feature/LQR_KF_Scheduler_altitude_control, feature/altitude_hold_control, feature/LQR_altitude_control, feature/kalman_filter_altitude` into `feature/Complete_altitude_hold_clean_code_version`.
- Not only is this an old branch `feature/simulations`, but as agreed we do not need the simulations in the repository. We can safely delete this. Worst case I have a local backup.
- I will delete `feature/43-angular_velocities_bias` now.

alphaville, 2023-08-30:
> * I agree merging `feature/LQR_KF_Scheduler_altitude_control, feature/altitude_hold_control, feature/LQR_altitude_control, feature/kalman_filter_altitude` into `feature/Complete_altitude_hold_clean_code_version`.
> 

These have been already merged into `feature/Complete_altitude_hold_clean_code_version`

gunturiCM, 2023-08-30:
Okay! I will delete them now.

closed 2023-09-05 by jamie-54