QUB-ASL/bzzz #110: Refuse to fly unless the switches are at the correct positions

alphaville, 2023-08-21:
Additional safety measure: The quadcopter should refuse to fly when the script starts unless the switches are at the correct positions (A: up, B: up, C: first position, D: down) **and** the throttle stick is fully down. In fact the buzzer should make a sound. Since we can control the buzzer only from the ESP, it should be good to implement this on the ESP.

alphaville, 2023-08-22:
@gunturiCM will you take care of this?

alphaville, 2023-11-29:
@jamie-54 @Yuanbwcx @Runway27 @pdavid747 @Minanchi We should take care of this issue before we start flying again to avoid accidents! Who can take care of this?

alphaville, 2023-11-30:
@jamie-54 @Yuanbwcx @Runway27 @pdavid747 @Minanchi folks, I still haven't received any replies from you on this. Who can take care of this issue?

jamie-54, 2023-11-30:
me and @Runway27 can look at this tomorrow

alphaville, 2024-01-18:
@jamie-54 and @Runway27 - should we close this issue?

Runway27, 2024-01-18:
Yes sir, this issue has been resolved. But the PR regarding this has issue #165 which is to be resolved.

alphaville, 2024-01-18:
Indeed, PR #172 is addressing two issues (#110 and #165), but this is not good practice. A PR should aim to close one issue (even when it partly addresses multiple ones). In any case, #172 still has unresolved conversations and has not been approved. 

I suggest we make PR #172 address only #110, we then review it quickly, we merge it, and we open a PR to work on #165. Does this work, @Runway27 and @jamie-54?

Runway27, 2024-01-18:
Yes sir, we can do that.

jamie-54, 2024-01-18:
> Indeed, PR #172 is addressing two issues (#110 and #165), but this is not good practice. A PR should aim to close one issue (even when it partly addresses multiple ones). In any case, #172 still has [...]

Sounds good

Runway27, 2024-01-29:
Are we happy to close this issue?

jamie-54, 2024-01-29:
> Are we happy to close this issue?
Yea sounds good to me

PR #172 (2024-01-25): Refuse to arm unless throttle stick is down and switches are at the correct positions
## Main Changes

The quadcopter will not arm unless all switches are up except the arm switch (switch B) and the throttle is less than 5%

## Associated Issues

Closes #110 

## Tests

Test that quadcopter will not arm
closed 2024-02-05 by jamie-54