QUB-ASL/bzzz #161: Acquisition and parsing of GPS data

alphaville, 2023-11-28:
### Describe the solution you'd like

I have uploaded the data we collected on 28/11/23 [https://gist.github.com/alphaville/995a15ef74e8d5b3128c87667b5bfdaf](here). This is how the data looks like

```
$GPGLL,5434.81291,N,00556.15593,W,145737.00,A,A*70
$GPRMC,145738.00,A,5434.81285,N,00556.15627,W,0.351,,281123,,,A*63
$GPVTG,,T,,M,0.351,N,0.650,K,A*27
$GPGGA,145738.00,5434.81285,N,00556.15627,W,1,05,1.28,64.4,M,52.5,M,,*79
$GPGSA,A,3,12,32,24,15,19,,,,,,,,2.51,1.28,2.16*02
$GPGSV,3,1,10,02,02,344,37,12,56,202,22,15,26,166,27,17,,,35*49
$GPGSV,3,2,10,19,34,071,36,22,11,050,32,23,18,236,,24,72,108,35*72
$GPGSV,3,3,10,25,21,223,,32,26,309,41*75
```

### Documentation

The data are structured according to the [NMEA Standard Logs](https://docs.novatel.com/OEM7/Content/Logs/NMEA_Standard_Logs_.htm) protocol. Examples:

1. [GPGLL](https://docs.novatel.com/OEM7/Content/Logs/GPGLL.htm) - Geographic Position
2. [GPGSV](https://docs.novatel.com/OEM7/Content/Logs/GPGSV.htm) - GPS Satellites in View
3. [GPVTG](https://docs.novatel.com/OEM7/Content/Logs/GPVTG.htm) - Track Made Good and Ground Speed

First of all we need to decipher what all these mean and how the data are structured (read the docs).

### Issues to address

1. We interface the GNSS over the serial at a baud rate of `500,000`. We can write a simple interface that reads the data from serial (using `pyserial`) and parses it
2. We should follow what we did [here](https://github.com/QUB-ASL/bzzz/blob/dev/raspberry/bzzz/sensors/anemometer.py) for the anemometer: a thread runs in the background, collects data and stores it in memory, and then we can access is with a non-blocking function (or get a moving average, median, etc). 
3. For some reason the data seems to be coming in at 1Hz, which is too slow. Can we speed this up?

alphaville, 2023-11-28:
@pdavid747 @Runway27 Note that if you write your own code you won't have to use `gpsd` at all. Actually you shouldn't use it because it won't let you read the serial. Start by reading the documentation to understand what sort of data we get from the GPS and then build a simple [DTO](https://en.wikipedia.org/wiki/Data_transfer_object): a Python class whose attributes are the quantities the GNSS measures.

alphaville, 2023-12-04:
### Accuracy of GPS measurements and conversion to degrees

⚠️ The GPS coordinates are in degrees-minutes-seconds; after the conversion to a decimal representation, it turns out that the coordinates are fairly accurate.

@pdavid747 use the following function to convert the GPGLL data to degrees:

```python
def deg_min_sec_to_decimal(dms, dir="N"):
    """
    Convert degrees-minutes-seconds from GPGLL to degrees (deciman)

    :param dms: degrees-minutes-seconds (must be positive)
    :param dir: direction (can be one of 'N', 'W', 'E', 'S')
    """
    dms_abs = np.abs(dms)
    degrees = np.floor(dms_abs/100)
    minutes = np.floor(dms_abs - 100 * np.floor(dms_abs/100))
    seconds = 10 * (dms_abs - np.floor(dms_abs))
    deg_decimal_abs = degrees + minutes / 60 + seconds / 3600
    dir_sign = -1 if (dir in ["S", "W"]) else 1
    return dir_sign * deg_decimal_abs
```

alphaville, 2023-12-04:
I installed PyGPSClient on Mac:

[image]

It allows to set the measurement update rate and many more settings.

alphaville, 2023-12-05:
Scatter plot of estimated positions while static:

[image]

while being connected to 3-8 satellites (the antenna is placed on metallic plate, but indoors having a decent but not perfect view of the sky; it has an elevation of around 75 degrees). We seem to be getting an accuracy of about ±10m at best, but we might get better results outdoors.

---

Some more results...

[image]
<p><em>Figure 1.</em> Satellites</p>

[image]
<p><em>Figure 2.</em> SNR</p>

[image]
<p><em>Figure 3.</em> Scatter plot (best one)</p>

[image]
<p><em>Figure 4.</em> Scatter plot</p>

| Parameter              | Value     |
| ------------------ | -------- |
| Satellites in view   | 7 - 9      | 
| <span title="Satellites in view">SIV</span> | 7-9 |
| <span title="Satellites in position solution">SIP</span> | 7 |
| <span title="Position dilution of precision">PDOP</span> | 3-3.5 |
| <span title="Horizontal dilution of precision">HDOP</span> | 1.5-3 |

alphaville, 2023-12-05:
### More data

From my other window with 7 satellites (10 in view), pdop: 1.97, and higher SNR

[image]
[image]
[image]

jamie-54, 2024-03-07:
Done

PR #201 (2024-03-05): Parse NMEA messages from GNSS module
This code parses GPS data and processes it like other sensors (data_logger/ filters) 

The code works by parsing serial GNSS data into decimal values of latitude longitude and then saves the data   to a csv file with date and time values per reading. 

## Associated Issues

- Closes #161 

## Tests

Has been tested on (Peters) MacBook