QUB-ASL/bzzz #164: RFC: Market research for GNSS modules and antennas

alphaville, 2023-11-30:
### Describe the solution you'd like

I'm looking for high-precision GNSS modules. These [ArduSimple simpleRTK2Blite Multiband GNSS Boards](https://www.mouser.co.uk/new/ardusimple/ardusimple-simplertk2blite-boards/) seems to be good candidates. They are priced at around £214 on Mouser. 

If we get a couple of modules, my understanding is that we can operate them in a base-rover configuration.

Just before I order this, can we check that (i) it's appropriate for our application, (ii) it's not too large and we're happy with the size, (iii) we've done a proper market research.

alphaville, 2023-11-30:
There's also this [SparkFun breakout](https://www.unmannedtechshop.co.uk/product/sparkfun-gps-rtk-sma-breakout-zed-f9p-qwiic/) with the same module, but seems to be smaller. It's just slightly more expensive (£233)

alphaville, 2023-12-01:
> https://www.mouser.co.uk/ProductDetail/SparkFun/GPS-16481?qs=vmHwEFxEFR8xw%2FiWbXmISw%3D%3D
> 
> I think this one would be best value for money as its refresh rate is only 5hz below the more expensive version, as well as that its well documented and we can use it with a usb port.

This uses RTK to give an update rate of 20Hz. We'd have to buy two of these (one for the quadcopter and one for the base station), but this is within our budget. 

Which is the more expensive version you're referring to?

pdavid747, 2023-12-01:
[SparkFun GNSS Combo Breakout - ZED-F9P, NEO-D9S](https://robosavvy.co.uk/sparkfun-gnss-combo-breakout-zed-f9p-neo-d9s-qwiic.html)

this one

pdavid747, 2023-12-01:
would we have to buy two of these also?

pdavid747, 2023-12-01:
if not it might be beneficial to go with that one as it will work out cheaper

alphaville, 2023-12-01:
> would we have to buy two of these also?

In the product description they say "With a clear view of the sky, specifically to the South, this combo breakout will deliver centimeter-level accuracy positioning without needing separate RTK or NTRIP corrections!" and "With this board, you will know where your (or any object's) X, Y, and Z location is better than an inch **without needing a separate correction source!**" 

I haven't found a comparison between the two modules, though. It would be also good to know whether it works reliable around our corner of the Earth. 

Just a thought: if we get two [SparkFun GPS-16481](https://www.mouser.co.uk/ProductDetail/SparkFun/GPS-16481?qs=vmHwEFxEFR8xw%2FiWbXmISw%3D%3D) we have two potential advantages: (i) we can get a NEO module later and connect it to the ZED-F9P, (ii) we'll have redundancy: if one module gets fried, most likely we'll still get decent accuracy and be able to fly (e.g., using SSR corrections).

pdavid747, 2023-12-01:
it illustrates the it is operational around the us and Europe, but its quite vague. I suppose if we found a video of someone using it in the uk we could find out that way

pdavid747, 2023-12-01:
https://www.youtube.com/watch?v=TSkIvtaVZc0

this video is quite useful

alphaville, 2023-12-01:
> https://www.youtube.com/watch?v=TSkIvtaVZc0
> 
> this video is quite useful

Good find! Indeed, [Mosaic X5](https://www.unmannedtechshop.co.uk/product/sparkfun-triband-gnss-rtk-breakout-mosaic-x5/) looks great, but it's at £610. However, it provides update rates at 100Hz and sub-cm errors.

alphaville, 2023-12-02:
Some additional information we need to be aware of:

This is what [SparkFun](https://learn.sparkfun.com/tutorials/what-is-gps-rtk) says about RTK:

> [...] an RTK receiver takes in an **RTCM correction stream** and then calculates your location with 1cm accuracy in real time. [...] You’ll also need a source of RTCM correction data. This usually comes from an internet connection or a long distance radio capable of approximately 500 bytes per second. LoRa and LTE-CAT M1 are superb choices for this backhaul.

More precisely we can get RTCM Corrections:

1. From [Skylark](https://www.swiftnav.com/skylark) but there's a monthly subscription of $50
2. By making our own NTRIP server that will provide RTCM corrections ([documentation here](https://learn.sparkfun.com/tutorials/how-to-build-a-diy-gnss-reference-station/all#introduction)); we could actually do that, but we'd need to get a second GNSS module and an antenna, but we can buy a cheaper GNSS module. 
3. From the [EUREF Permanent GNSS Network](http://www.epncb.oma.be/_networkdata/data_access/real_time/map.php) and we're lucky that there's a station in Belfast somewhere at the harbour

In anycase, unless we need to fly far away from Belfast (>10km from harbour), we won't need to make our own NTRIP server.

### So, which one do we buy?

[Sparkfun Mosaic X5](https://www.unmannedtechshop.co.uk/product/sparkfun-triband-gnss-rtk-breakout-mosaic-x5/) still seems like a good idea, but without RTCM corrections the accuracy will be poor (around ±1.5m). However, we can get correction data from the Belfast station. The only downside is that an Internet connection is required, but I've already ordered a 4G-enabled WiFi router, so we'll have a connection anywhere we go.

@pdavid747, @jamie-54, @Runway27, @Yuanbwcx, @Minanchi thoughts?

alphaville, 2023-12-02:
**Update:** The Belfast NTRIP station has been down since April 2023 ([link](http://www.epncb.oma.be/_networkdata/siteinfo4onestation.php?station=BFST00GBR)). The closest active RTK is in [Ballymena](http://monitor.use-snip.com/map).