QUB-ASL/bzzz #171: correct paths for data logger and filters

jamie-54, 2023-12-05:
### Describe the bug

Fails to run `python bzzz/raspberry/main.py`

This is due to the path for importing `DataLogger` and `MedianFilter`/`NoFilter` in the `anemometer.py` and `evo_time_of_flight.py` files.

However if I add `bzzz.sensor.` it will run the `main.py` file but then the `anemometer.py` and `evo_time_of_flight.py` files

### Affected File

bzzz/raspberry/bzzz/sensor/anemometer.py 

### Minimum reproducible example

```

```

alphaville, 2024-01-18:
I think local imports (when importing a module into another in the same folder) need a dot at the beginning. For example, in [another project] we have the folder `constraints` and there we have the files [`ball1.py`](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/opengen/constraints/ball1.py) and [`simplex.py`](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/opengen/constraints/simplex.py). To import `simplex` into `ball1` [we do](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/opengen/constraints/ball1.py#L2):

```py
from .simplex import Simplex
```

In fact, we use `from ... import ...` to import only the things we need from `simplex.py`.

In [another case](https://github.com/alphaville/optimization-engine/blob/master/open-codegen/opengen/constraints/finite_set.py#L4), we wanted to import a whole sub-module, so we did:

```py
import opengen.functions as fn
```

Here `opengen` is the name of the top-level model (ours is `bzzz`).

alphaville, 2024-01-18:
@jamie-54 do we have a PR for this?

jamie-54, 2024-01-18:
> @jamie-54 do we have a PR for this?

No, there isn't a branch for this either