QUB-ASL/bzzz #175: RFC: killing and un-killing safety procedure

jamie-54, 2024-01-19:
### Describe the solution you'd like

We're about to merge #172 where to arm the throttle stick needs to be down. Now as we fly, if we panic and kill it and then land, the throttle stick will be not at the zero position. Then, to fly again we'll un-kill it and... boom! At the same time, if while flying, we kill it by mistake we want to be able to un-kill immediately - without start-up routines. Here's an idea: we introduce a timer (~3s) and **once killed for 3s, the throttle stick has to be at zero to un-kill**.

alphaville, 2024-01-22:
@jamie-54 I guess this can be addressed in Python, right? We shouldn't have to change anything in the ESP32 code.

jamie-54, 2024-01-22:
At the minute part of its addressed on the ESP32 if you are free I can show you

closed 2024-02-05 by jamie-54