QUB-ASL/bzzz #188: Save EVO TOF sensor data when flying

Yuanbwcx, 2024-01-30:
### Describe the solution you'd like

Be able to save Evo TOF data to check if the sensor work while flying

jamie-54, 2024-02-08:
@alphaville @Yuanbwcx I think this issue was just to check that the EVO time of flight sensor works while flying. This has been tested and works. I can push the branch and make a PR but I'm not sure if it will be worth merging as it is only a few lines of code that'll need changed anyway?

Maybe we should make a new issue outlining what @alphaville has mentioned above?

Yuanbwcx, 2024-02-08:
@alphaville Yes, sir I will take care of it. Are we gonna replace the old sensor or use both at the same time?

alphaville, 2024-02-08:
@Yuanbwcx as a first priority I would simply replace the old sensor with the new one.

alphaville, 2024-03-01:
This is being addressed in #194

jamie-54, 2024-03-02:
Should we close this?

jamie-54, 2024-03-07:
Done

PR #202 (2024-03-07): Initiate threading on main file
## Main Changes

The threading now begins on the main file. 

## Associated Issues

- Closes #196 
- Closes #188 
- Closes #186

## Tests

Has been tested on the quadcopter and all the sensors are collecting the data.
closed 2024-03-07 by jamie-54