QUB-ASL/bzzz #196: Initiate threading from main.py

Runway27, 2024-02-22:
### Describe the solution you'd like

We need to initiate threading for the barometer from the main file.

alphaville, 2024-03-01:
@Runway27 Can you take care of this issue? It should be rather straightforward (just import the barometer class and use it)

Runway27, 2024-03-01:
Yes sir. I have already created a branch. And have started working on it. Hopefully it will be completed on Monday.

Runway27, 2024-03-01:
I’m planning to include both barometer and time of flight for the time being. And we can include GPS later.

alphaville, 2024-03-01:
It should be really straightforward. Have a look at [multiple with statements](https://stackoverflow.com/questions/893333/multiple-variables-in-a-with-statement).

Runway27, 2024-03-01:
Yes sir.

jamie-54, 2024-03-02:
> I’m planning to include both barometer and time of flight for the time being. And we can include GPS later.

Include the anemometer as well please, [this](https://github.com/QUB-ASL/bzzz/blob/feature/186-save_wind_data_when_flying/raspberry/main.py) should help

jamie-54, 2024-03-07:
Done

PR #202 (2024-03-07): Initiate threading on main file
## Main Changes

The threading now begins on the main file. 

## Associated Issues

- Closes #196 
- Closes #188 
- Closes #186

## Tests

Has been tested on the quadcopter and all the sensors are collecting the data.
closed 2024-03-07 by jamie-54