QUB-ASL/bzzz #200: Mount GNSS and Antenna to bzzz

jamie-54, 2024-03-01:
### Describe the solution you'd like

* [ ] Mount GNSS Module to lid
* [ ] Redesign anemometer case or design a platform to mount the antenna to the carbon fibre tube

I've ordered the [USB cable](https://www.amazon.co.uk/charge-Cable-MYLB-Braided-Charging-Black/dp/B07VCHVH9K?source=ps-sl-shoppingads-lpcontext&ref_=fplfs&smid=ABZK44ZAKI0T3&th=1) to connect the GNSS Module to the Pi

jamie-54, 2024-03-01:
I'm going to order [this connector](https://uk.rs-online.com/web/p/coaxial-connectors/4466432), to make it easier to mount the antenna

jamie-54, 2024-03-01:
I was thinking something like this for anemometer case with antenna stand?
What's your thoughts?
[image]

jamie-54, 2024-03-01:
or maybe like this?

[image]

alphaville, 2024-03-08:
@jamie-54 will there be some holes through the top plate? How will the antenna be supported?

jamie-54, 2024-03-08:
> @jamie-54 will there be some holes through the top plate? How will the antenna be supported?

Hi, I haven't include holes as they where going to line up along the split in the case and it couldn't really be rotated because of the slot for the cable. I was thinking the double sided tape would be enough to hold it as the antenna is very lightweight?
I have this one printed and on the quadcopter and it looks like it will work well

closed 2024-03-11 by jamie-54